kplindegaard/smbus2 #4: Python3

howroyd, 2016-09-28:
Will this work with python3.5? Many thanks

kplindegaard, 2016-09-28:
I must admit that I haven´t used smbus2 on Python 3.5. Only on various Raspberry Pis running Python 2.7. 
However, I have mocked the underlying IO library (the ioctl-function in fcntl) as much as practically possible, and those run just fine Travis on Python 3.5 (https://travis-ci.org/kplindegaard/smbus2). Basically, if the ioctl implementation on your target machine behaves, so should smbus2. 

All in all: I am quite confident that smbus2 will work like a charm on all Python version from 2.7 and up to 3.5, but I don´t have the hard evidence to back that statement up with. I would of course be very interested to hear from you and your experiences.

kplindegaard, 2016-11-21:
Closing the issue. I have tested with a completely clean Python 3.4. Can't see that Python 3.5 will behave differently.

closed 2016-11-21 by kplindegaard