kplindegaard/smbus2 #6: Function Call

PvanHengel, 2016-12-21:
Good Morning,

I am trying to preform a function call, as described in the doc https://www.pozyx.io/Documentation/Datasheet/Interfaces but am not able to get it to work.   I am using th PI platform, and had to turn on "combined" mode, by setting the file to Y sudo nano /sys/module/i2c_bcm2708/parameters/combined and that makes the reads work as expected.  Do you have any tips how to make this function call using the library. 

Thank you!
Peter

PvanHengel, 2016-12-22:
Hi yes I have those settings set.  I think the issue I am seeing is that I can read & write, but cant to a process call:

SMBus Process Call:
===================

This command selects a device register (through the Comm byte), sends
16 bits of data to it, and reads 16 bits of data in return.

S Addr Wr [A] Comm [A] DataLow [A] DataHigh [A] 
                             S Addr Rd [A] [DataLow] A [DataHigh] NA P

Functionality flag: I2C_FUNC_SMBUS_PROC_CALL

(https://www.kernel.org/doc/Documentation/i2c/smbus-protocol)

Have you been able to get this function working.  I have tried a write, followed by a read, but I dont think the sequence is quite right.

P

kplindegaard, 2016-12-23:
I misunderstood you. Currently process call is not implemented. 

It requires a new method in the SMBus class. Unfortunately none of my devices support this, so I can't really test it in practice. All I can offer is that I have a go at it at you test it for me?

PvanHengel, 2016-12-23:
Cool that would be great it seems that reads work fine but writes do not.  How do we control repeated starts?

P

PvanHengel, 2016-12-23:
I would be happy to test or provide ssh instructions.  My concern now is that I can't get writes to work that trying to get a function call working when writes don't work is difficult

kplindegaard, 2016-12-23:
It's a bit suspicious that write doesn't work for you. You do mean normal write operations like bus.write_byte_data() and the like? I suspect that if you can't get the normal write ops to work, it's highly unlikely that a process call will behave OK.

Anyway. I have branched off a new branch named "feature/processcall" where a the new process call method is now available on the SMBus object. See Readme.md or Readme.rst in the commit 9d405f93651988cc84de4eecfdf3793aded1284d for how to use it.

Recipe: Clone the repository, and checkout that branch. Let me know if it works or not. I.e.
git clone https://github.com/kplindegaard/smbus2.git
cd smbus2
git checkout feature/processcall

PvanHengel, 2016-12-28:
Hi, I'm not sure I am able to test, as that I still can't get write to work.   I am using https://www.pozyx.io/Documentation/Datasheet/Interfaces as my sensor.  I am able to read fine, as long as I enable repeated starts.  The write, however does not seem to be a pure smbus write as you can see in the doc.  I cant figure out how using this library to get it to put that exact sequence in.  I'm testing using register 0x15 to control the LED's.   I think the issue is that it is not sending the repeated start.  I also looked into another library (https://github.com/vsergeev/python-periphery/blob/master/docs/i2c.rst) but cant seem to get it to work either.   Is there a way to have more control?

PvanHengel, 2016-12-29:
> Have you tried the write_i2c_block_data() function?

Yes I have that doesn't seem to work.  I think it has to do with the repeated start not being sent for writes only reads.

kplindegaard, 2016-12-30:
I'm not really an expert when you talk low-level timings and those things, but as you certainly already know by now repeated start has been (and for all I know still is) an issue on the Pi.
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=15840

It could be you have to look for another lib than smbus2 if your device requires repeated start. As of now I don't see a way to configure this from pure Python without embedding custom C/C++ code in there. Sorry.

kplindegaard, 2016-12-30:
Googling some more, there is apparently a workaround/fix for this. Check http://prng.blogspot.no/2014/10/raspberry-pi-and-lack-of-i2c-repeated.html and see if the suggestion in the comment section works for you.

PvanHengel, 2016-12-30:
Thank you I think this is related, I came across this before as reads were not working at all every read returned 0x00 after changing the file to combined mode all reads work fine however writes do not.  

I don't have an oscilloscope nor do I know which one to get and how to use so that is making this much more difficult to debug.

kplindegaard, 2017-01-10:
I came across an i2c library that may come to your rescue, namely 
[quick2wire's i2c Python library](https://github.com/quick2wire/quick2wire-python-api)
It was developed for Python 3.x, and Python 2.x support seems to be a pending pull-request.

You could try to use modified version, the source for the pull request: [freason's python 2 branch](https://github.com/freason/quick2wire-python-api/tree/support_python2) that supports both Python 2.x and 3.y.

kplindegaard, 2019-01-09:
I'm closing this issue now, because smbus2 for two reasons:

- In the next version we will have SMBus `process_call`, and
- Users already have the possibility since v0.2.0 to also craft i2c combined write and read transactions. You know, with "repeated start".

We are (I think...) at the end of the road on this matter.

closed 2019-01-09 by kplindegaard