kplindegaard/smbus2 #15: install error

SuReinfelder, 2017-12-01:
Hello,
I just cloned the repo today and I used the command "python setup.py install" and then it told me "traceback, file "setup.py" line 1 in <module> from setuptools import setup ImportError: no module named setuptools".
I am new to using python and I am new to GitHub as well so it might be possible I just missed something. But I desperately need your nice i2c block write/read routines and I would be really glad if you could point me to the right direction.
Thanks in advance!
Kind regards
Susanne

kplindegaard, 2017-12-01:
Hi,
Maybe you have already found a solution? If not, I would recommend to [install pip.](https://pip.pypa.io/en/stable/installing/). Check it out.

It's incredibly convenient to be able to explore, download and install whatever you need with just a single command. For example:
```
pip install smbus2
```

SuReinfelder, 2017-12-04:
Hi,
unfortunately I didn't find a way to use
`python setup.py install`

Following your advice revealed that, although I had python installed in version 2.7.9, `pip` wasn't installed.
After installing `pip` and executing `pip install smbus2` it works now.

Maybe this is some kind of a "newby on linux" bug. Sorry for bothering you here!

Kind regards
Susanne

closed 2017-12-04 by SuReinfelder