kplindegaard/smbus2 #19: Read 2 bytes without register / offsets

PiStuffing, 2018-03-31:
This is likely to be my problem that I need help with.  I'm using this i2c joysticks:

http://www.grayhill.com/assets/1/7/67a_i2c_user_manual_1.2.pdf

It requires reading 2 bytes simultaniously; there is no register used.  Python SMBus doesn't support this function, hence I'm trying your SMBus2.  Of the four options I've tried, only the first works (a bit but only one byte), the rest either get exceptions or return all zeros.  Any suggestions please?

`    with SMBusWrapper(1) as bus:

        bus.write_byte_data(0x40, 0x76, 2)
        bus.write_byte_data(0x41, 0x76, 2)
        
        while True:
            time.sleep(0.5)

            # data = bus.read_byte(0x40)
            # data = bus.read_byte_data(0x40, 0)
            # data = bus.read_word_data(0x40, 0)
            data = list(i2c_msg.read(0x40, 2))
            # data = bus.read_i2c_block_data(0x40, 0, 2) 

            assert (len(data) == 2), "Joystick 0 data len: %d" % len(data)
            print "0x40: ",
            print data,`

kplindegaard, 2018-04-01:
Just flipping through the pdf you referred to, I wouldn't expect any of these to work:
* read_byte_data()
* read_word_data()
* read_i2c_block_data()

I have two comments/questions for you:
1. Have you tried two repeated `bus.read_byte(0x40)`? It's a bit rough, but it has worked for me for one particular i2c device in the past. Depends on how forgiving yours is.
1. You could also try the `bus.i2c_rdwr()`, and I suppose that's what you intended to do. However, you are not doing it right as far as I can see in the code sample.

```
# Read 2 bytes from address 0x40
msg = i2c_msg.read(0x40, 2)
bus.i2c_rdwr(msg)

# Access the data in the msg object
data = list(msg)
print(data) 
```

Good luck

PiStuffing, 2018-04-01:
Perfect, thank you.  Problem solved.

While two read_bytes returned the same value on both reads, your i2c_msg + i2c_rdwr are producing 2 distinct values tracking the joysticks perfectly.

closed 2018-04-01 by PiStuffing