kplindegaard/smbus2 #21: error in smbus2 setup command: 'extras_require'

ash73, 2018-07-13:
Hi, trying to install smbus2 on a rPi2, because something else I am trying to install(!) may have a dependency on it, but smbus2 install fails:

sudo python setup.py install

error in smbus2 setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.

I had a look in setup.py and it says:
    extras_require={
        'docs': [
            'sphinx >= 1.5.3'
        ],
        'qa': [
            'flake8'
        ],
        'test': test_deps
    }

But I have no idea how to fix it.

Please help.

kplindegaard, 2018-07-13:
Works like a charm on my Linux box with Python 2.7.13 and setuptools 36.4.0.

You can try to delete the `extras_require` and install. It's not strictly necessary for installing the core of the lib.

closed 2018-07-19 by ash73

tedder, 2018-08-04:
Dead ticket, but maybe it's a py3/pip3 problem.

ash73, 2018-08-04:
My Raspbian install must have been corrupted somehow, I created a fresh install and everything works ok.