kplindegaard/smbus2 #22: Gracefully handling trackbacks?

AlexKlimaj, 2018-08-22:
Is there a way to gracefully handle this in a program?

This happens very rarely in my code, but when it does, it exits a long running test. 

```
Traceback (most recent call last):
  File "battery_test.py", line 89, in <module>
    remaining = bus.read_byte_data(0x0b, 0x0d)
  File "/usr/local/lib/python3.7/site-packages/smbus2/smbus2.py", line 355, in read_byte_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error
```

AlexKlimaj, 2018-08-22:
Solved by making higher level functions with error handling. 

`
def battery_block_read(address, register, numberofbytes):
    tmp = None
    while (tmp == None):
        with SMBusWrapper(1) as bus:
            try:
                tmp = bus.read_i2c_block_data(address, register, numberofbytes)
                if (tmp != None):
                    return tmp;
            except:
                print("SMBUS read block error")
    return;
`

closed 2018-08-22 by AlexKlimaj

kplindegaard, 2018-08-23:
Yep, you’ll need to add error handling yourself. The philosophy is to keep the library as small as possible :)