kplindegaard/smbus2 #25: How to do repeated starts

chetferry, 2018-12-19:
I am attempting to do the following with a Raspberry pi zero w:

start condition

address+write bit
write byte

start condition

address+read bit
read byte
read byte
read byte

stop condition

Can you suggest the right combination of calls using the smbus2 library?

I am successfully talking on the bus.  I can see with my logic analyzer the device ACK.  I have failed thus far to figure out how to do the write one byte, send another start then a read command then read the 3 bytes.  I can do all kinds of other communication just fine using the smbus2 library.  Thanks in advance for any help that could be provided.

kplindegaard, 2018-12-19:
Have you tried setting up an "i2c_rdwr" combination similar to one in example 6? 

```
from smbus2 import SMBus, i2c_msg

# Transaction writing one byte then reading three bytes
addr = 80
output_byte = 45  # Byte value to write
write = i2c_msg.write(addr, [output_byte,])
read = i2c_msg.read(addr, 3)
with SMBusWrapper(1) as bus:
    bus.i2c_rdwr(write, read)

# Convert read message to a list and print it
data = list(read)
print(data)
```

kplindegaard, 2019-01-02:
@chetferry any updates?

kplindegaard, 2019-03-04:
Recipe given so I'm closing this one.

closed 2019-03-04 by kplindegaard