kplindegaard/smbus2 #26: smbus2 setup failure: 'extras_require'

MariusTOR, 2019-01-01:
I am trying to install smbus2-0.2.1 on a Pocketbeagle running Debian v9.5.  While I do not have an internet connection, I've utilized WinSCP to get the appropriate wheels and tar files over to the beagle to install pip v18.1 and setuptools v40.6.3.  However, upon trying to install smbus2-0.2.1, I get the following error:

```
debian@beaglebone:/var/lib/cloud9$ pip install smbus2-0.2.1.tar
Processing ./smbus2-0.2.1.tar
    Complete output from command python setup.py egg_info:
    error in smbus2 setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-f0O3p9/
```

Per a previous bug report (#21 ), I tried untarring smbus2-0.2.1 and commenting out the "extras_require" lines.  Upon running setup.py, I got the following:

```
debian@beaglebone:/var/lib/cloud9/smbus2-0.2.1$ python setup.py
Traceback (most recent call last):
  File "setup.py", line 63, in <module>
    "Programming Language :: Python :: 3.6"
  File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
    _setup_distribution = dist = klass(attrs)
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/setuptools/dist.py", line 225, in __init__
    _Distribution.__init__(self,attrs)
  File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
    self.finalize_options()
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/setuptools/dist.py", line 257, in finalize_options
    ep.require(installer=self.fetch_build_egg)
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/pkg_resources.py", line 2072, in require
    working_set.resolve(self.dist.requires(self.extras),env,installer))
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/pkg_resources.py", line 2282, in requires
    dm = self._dep_map
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/pkg_resources.py", line 2511, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/pkg_resources.py", line 2544, in _compute_dependencies
    common = frozenset(reqs_for_extra(None))
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/pkg_resources.py", line 2541, in reqs_for_extra
    if req.marker_fn(override={'extra':extra}):
  File "/usr/local/lib/python2.7/dist-packages/distribute-0.6.45-py2.7.egg/_markerlib/markers.py", line 109, in marker_fn
    return eval(compiled_marker, environment)
  File "<environment marker>", line 1, in <module>
NameError: name 'sys_platform' is not defined
```
I'm not sure where to go at this point.  Previous solutions were based on upgrading the version of setuptools that was available, but I have the most up-to-date version, so that's out.  Any thoughts?

kplindegaard, 2019-01-02:
Installing with `python setup.py install` after downloading v0.2.1 either as [.tar](https://github.com/kplindegaard/smbus2/archive/0.2.1.tar.gz) or [.zip](https://github.com/kplindegaard/smbus2/archive/0.2.1.zip) works like a charm for me with Python 2.7.15 and setuptools 40.6.3 when I comment out the extras_require dict in setup.py.

Check two things:

1. Seems like you run `python setup.py` and not `python setup.py install` as the installation recipe suggests...?
1. Sure that you commented the extras_require part correctly?

Silly questions, sorry, but I'm not an expert debugging this. If you still have problems please provide your modified setup.py file.

MariusTOR, 2019-01-03:
> Maybe I misunderstood where you got the .tar file from?

Ah, you got it.  I just didn't read through the readme like I should
have...everything is working well.  Thanks!

kplindegaard, 2019-01-03:
@MariusTOR great stuff. Closing this.

closed 2019-01-03 by kplindegaard