kplindegaard/smbus2 #33: best way to use SMBusWrapper

bbbart, 2019-03-01:
Hi,

I've been using your library for a while on a Pi talking to four Arduino's over I2C.

Usually(™) things go fine, but from time to time the whole bus seems to get messed up and no writes/reads are succeeding any more.

My program fires off a thread which basically has a `while True` loop in which it reads/writes from/to i2c whenever instructed to. This often results in up to four read/write instructions per second during one hour.

How would you suggest to use SMBusWrapper in this scenario?

I am currently wrapping the `with` statement inside the loop, but cold also run the loop inside the `with`, I suppose. Then I don't open/close a the bus four times per second.

Would this make any difference, in your opinion?

Thanks a lot!

kplindegaard, 2019-03-02:
My use cases are a little different from yours. I've never had more than one slave hooked up on the same device. Mine involves batches of read and write operations to one single i2c slave. Can be up to  10+ operations every time with each read/write enclosed in a with-clause. What I have seen though, is that the type of slave device I use the Pi with often need a little time between each r/w command. However, I've never had any problem like you are reporting.

As far as I know, both your options should be OK. I guess you just have to try both and see which is better. Alternatively also experiment with introducing a short delay between each read/write if that's possible.

bbbart, 2019-03-04:
Hi,

It was Sat 02 Mar 2019 at 03:31 when Karl-Petter Lindegaard wrote:
> My use cases are a little different from yours. I've never had more
> than one slave hooked up on the same device. Mine involves batches of
> read and write operations to one single i2c slave. Can be up [...]

I see. I currently have four Arduino's hooked up as slaves to one
Raspberry Pi master. :-)

> As far as I know, both your options should be OK. I guess you just
> have to try both and see which is better. Alternatively also
> experiment with introducing a short delay between each read/write if
> that's possible.

I think I solved the problem. Hopefully... The reason the application
stopped reading/writing from/to the i2c bus had nothing to do with
smbus2. I'm using this in a thread that reads/write i2c bytes from/to
SimpleQueue objects, and something completely unrelated was causing that
thread to finish prematurely.

Since setting up the bus seems quit inexpensive an operation, I'm
leaving it inside the while loop for now.

Thanks for your reply!

-- 
regards,
Bart Van Loon
 
I suppose the secret to happiness is learning to appreciate the moment.
		-- Calvin

kplindegaard, 2019-03-04:
Great that you found the underlying problem!
 Agree that setting up the bus is a inexpensive operation, which is why I get away with opening and shutting it down 10+ times in my app. No need to keep it open if you don't strictly need to.

closed 2019-05-11 by kplindegaard