kplindegaard/smbus2 #35: read requests when over I2C_SMBUS_BLOCK_MAX

cyphunk, 2019-03-12:
I'm not entirely certain the reason for the I2C_SMBUS_BLOCK_MAX being 32 bytes and curious if I could investigate how to modify this. If you have suggestions of what requires this hard limit please let me know. I have some time to investigate. Currently trying to emulate an i2C touch screen that results in reads well over 32 bytes

thanks

kplindegaard, 2019-03-12:
@cyphunk: smbus2 uses the drivers provided by the Linux kernel, and it's kind of difficult to bypass if you insist to use the `read_i2c_block_data` method. Explanation given for example [here](https://www.kernel.org/doc/html/v4.9/driver-api/i2c.html#c.i2c_smbus_read_block_data). Right off the bat I don't remember if that limitation is actually there in the [SMBus specification](http://www.smbus.org/specs/), but it doesn't really matter as long as we're limited by the kernel driver's struct definition anyway.

However, doing i2c reads in larger chunks than 32 bytes is not a problem. Straight forward if don't need to provide an offset address where to start reading from, as a matter of fact. Just check out example 5 and 7 in the project readme.

cyphunk, 2019-03-21:
I think indeed you are right about having to dig into smbus. I banged my head against pigpio (https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=236351). If I have another week I may dig further into smbus

thank you for the references!

closed 2019-03-21 by cyphunk