kplindegaard/smbus2 #37: Trying to interface MMC3416xPJ sensor with smbus2

sdrck, 2019-05-05:
Dear @kplindegaard,

Thank you for all your effort and your very nice smbus2 interface for python, which will hopefully help me with my problem to interface a 3d magnetic field amr sensor. Unfortunately, I don't really know, whether it is possible to use your package to solve my problem. Is smbus2 using i2c_rdwr able to distinguish between first and repeated starts? Sorry, but I didn't really get this from your documentation / examples and I hope that you could just give me a hint, how I can get the sensor running. I already opened a discussion showing by C code I want to port to python on stack overflow: 

https://stackoverflow.com/questions/55987807/how-to-communicate-with-mc3416xpj-via-pythons-smbus-i2c-interface

I really would appreciate if someone could help me. Thank you in advance.

Kind regards!

kplindegaard, 2019-05-11:
@sdrck maybe you've already tried since your question is already 6 days old?

In smbu2s you don't deal with the start and stop bits like in the C example. However, the `i2c_rdwr()` method sends a *repeated start* in between each `i2c_msg` object payload you provide as inputs to it. The C example you provide is a bit like example 6 in the readme. Try to reconstruct it with a series of writes and reads you pass in to `i2c_rdwr` and see where that takes you.

wschmied, 2020-06-07:
@sdrck If you are looking for an implementation with repeated starts: https://github.com/madewhatnow/TFmini-I2C-Python/blob/master/TFmini_I2C.py is for an I2C sensor. The sensor documentation (including the read/write sequence) are in the pdf in the repository, maybe that helps with finding a solution for yours?