kplindegaard/smbus2 #38: Help with a software reset command

rossmacarthur, 2019-07-11:
My I2C chip has a software reset command.

[image]

How could I implement this using smbus2? I have no problem writing to the 0x40 address using

```python
from smbus2 import SMBusWrapper

with SMBusWrapper(0) as bus:
    bus.write_byte_data(0x40, 0x0C, 0x01)
```

kplindegaard, 2019-07-20:
@rossmacarthur: 
The recipe seems to match the i2c write diagram halfway down the [i2c specification](https://i2c.info/i2c-bus-specification).

[image]

What I suggest is that you modify Example 5 in the readme and supply a triplet consisting of the SWRST call, Byte 1 and Byte 2 as payload:
```
from smbus2 import SMBus, i2c_msg

with SMBusWrapper(0) as bus:
    # Write SWRST payload to address 1001 011
    adr = 0x4b  # 1001 011
    payload = [0x96, 0xa5, 0x5a]
    msg = i2c_msg.write(adr, payload)
    bus.i2c_rdwr(msg)
```
See if that does the trick.

rossmacarthur, 2019-07-29:
Hmmm unfortunately it doesn't work

```
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
  File "venv/local/lib/python2.7/site-packages/smbus2/smbus2.py", line 497, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
IOError: [Errno 6] No such device or address
```

kplindegaard, 2019-07-29:
So what does `i2cdetect -y 0` tell you?

rossmacarthur, 2019-07-29:
```
$ i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- 4b -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

kplindegaard, 2019-07-29:
So addr 0x4b is there. Good.
But honestly, I don't see anything wrong with the payload `[0x96, 0xA5, 0x5A]` ... :/