kplindegaard/smbus2 #43: deteced i2c adress

KJPOUNDY132, 2020-01-10:
hi how can i list avaible i2c adresses in python code

kungpfui, 2020-01-13:
Trial and error is your friend.

I wrote a small snippet of code which shows similar results as _i2cdetect_ do.
[i2c device scan example](https://gist.github.com/kungpfui/54784ebc3b3ca72169c1839720b313bf)

Cheers
Stefan

KJPOUNDY132, 2020-01-15:
Thanks guy

Windows 10 için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

________________________________
Gönderen: Stefan S. <notifications@github.com>
Gönderildi: Monday, January 13, 2020 11:23:39 PM
Kime: kplindegaard/smbus2 <smbus2@noreply.github.com>
Bilgi: KJPOUNDY132 <yocciomala@outlook.com>; Author <author@noreply.github.com>
Konu: Re: [kplindegaard/smbus2] deteced i2c adress (#43)

Trial and error is your friend.

I wrote a small snippet of code which shows similar results as i2cdetect do.
i2c device scan example<https://gist.github.com/kungpfui/54784ebc3b3ca72169c1839720b313bf>

Cheers
Stefan