kplindegaard/smbus2 #48: Read smbus 32 byte Block data

scottscott70, 2020-07-03:
I use read_i2c_block_data function to read bq40z55 data flash
but once block_data have 36 Byte, i2c_block_data function limit to 32 Byte
how can I do read all 36 Byte Block data
[image]

kplindegaard, 2020-07-03:
Hi @scottscott70 
Take a look at [Example 5](https://github.com/kplindegaard/smbus2#example-5-single-i2c_rdwr) in the readme.

scottscott70, 2020-07-04:
Thanks for your reply.
After the test, the i2c waveform was correct.
But how to get data buffer?
Now , I get data buffer is nano.
This smbus2 module was perfect. Thank You.

[image]
[image]

scottscott70, 2020-07-04:
Buy the way, "pip install smbus2" is not working Now.

scottscott70, 2020-07-04:
Sorry, I got data succesfully

[image]
[image]

closed 2020-07-06 by scottscott70