kplindegaard/smbus2 #58: about read_block_data(i2c_addr, register, force=None)

mskkk, 2020-12-21:
I use i2c to connect rpi and arduino . I check the data i sent to arduino in Serial Monitor if i  sent a array by write_block_data() function , arduino receive start register + length + array .
I want to know how to use read_block_data() to read the data returned by arduino .
I tried these days , rpi always receive a empty list  and arduino Wire.write() only send back a byte data how can i combine these two function together.

kplindegaard, 2021-01-17:
Afraid I can't help you much in your quest. I have no Arduino experience myself.

While looking for the answer, you might wanna check two things:
1. The i2c driver support for the command you are using from the Pi.

```python
import smbus2

# Assuming you already have opened the `bus`
rbd_supported = bus.funcs & smbus2.I2cFunc.SMBUS_READ_DATA > 0
print(rdb_supported)
```
If you don't see `True` here, the Pi's I2C driver doesn't support the operation.

2. Another source of error could be the Ardiuno. Given my limited experience, the best guess would be to check how the Wire lib works and if you need to alter its response to an incoming read block command.

grandaphi, 2021-07-08:
I am having a problem with read_block_data too. I tried this but SMBUS_READ_DATA appears not to be an attribute of smbus2.I2cfunc:

```
>>> import smbus2
>>> dir (smbus2.I2cFunc)

['ADDR_10BIT', 'I2C', 'NOSTART', 'PROTOCOL_MANGLING', 'SLAVE', 'SMBUS_BLOCK_DATA', 'SMBUS_BLOCK_PROC_CALL', 'SMBUS_BYTE', 'SMBUS_BYTE_DATA', 'SMBUS_EMUL', 'SMBUS_HOST_NOTIFY', 'SMBUS_I2C_BLOCK', 'SMBUS_PEC', 'SMBUS_PROC_CALL', 'SMBUS_QUICK', 'SMBUS_READ_BLOCK_DATA', 'SMBUS_READ_BYTE', 'SMBUS_READ_BYTE_DATA', 'SMBUS_READ_I2C_BLOCK', 'SMBUS_READ_WORD_DATA', 'SMBUS_WORD_DATA', 'SMBUS_WRITE_BLOCK_DATA', 'SMBUS_WRITE_BYTE', 'SMBUS_WRITE_BYTE_DATA', 'SMBUS_WRITE_I2C_BLOCK', 'SMBUS_WRITE_WORD_DATA', '__class__', '__doc__', '__members__', '__module__']
```

For me the issue with read block data is it is returning the correct number of bytes however they are all zeros, which should not be the case

kplindegaard, 2021-07-08:
Sorry. Was a typo in my previous comment. Should have been `SMBUS_READ_BLOCK_DATA`, meaning that the example should go like this:

```python
import smbus2

# Assuming you already have opened the `bus`
rbd_supported = bus.funcs & smbus2.I2cFunc.SMBUS_READ_BLOCK_DATA > 0
print(rdb_supported)
```

RPi's normally don't support `read_block_data()` so you need to find an alternative. For example `read_i2c_block_data()`.

grandaphi, 2021-07-09:
thanks, yeah it returns `False` . 
Just fixed up another typo for anybody else who wants to use this. `rdb_supported`

`import smbus2

# Assuming you already have opened the `bus`
rdb_supported = bus.funcs & smbus2.I2cFunc.SMBUS_READ_BLOCK_DATA > 0
print(rdb_supported)`