kplindegaard/smbus2 #63: Witting message with length > 32 to trigger device functionality

mefistotelis, 2021-01-23:
Here is a link do an image taken from spec of a device which uses SMBus communication (BQ series chip from TI).
https://github.com/o-gs/dji-firmware-tools/issues/17#issuecomment-766040778

Note that the use of specific function of the device (read firmware version) requires to write block_data with size=62.
(actual data is never sent, as the device responds with NACK on the size)
Currently, the smbus2 implementation will prevent sending such packet, making it impossible to use this functionality of the BQ series chips.

The restriction should be lifted, or another lower level API should be prepared.

mefistotelis, 2021-01-23:
Wait, isn't that pull request exactly what is needed to solve this problem?

https://github.com/kplindegaard/smbus2/pull/23

kplindegaard, 2021-02-02:
Already supported. 32 byte max buffer length are only apply to the SMBus operations and not i2c. Please see the [I2C](https://github.com/kplindegaard/smbus2#i2c) section of the readme.

The `read` and `write` methods introduced in that PR are just for convenience, and as per my comment to that PR it's still unfinished business associated with and not going in as it stands.

mefistotelis, 2021-02-02:
You are right. I should have read the readme.
(though these convenience functions from PR are very convenient)

Thank you!

closed 2021-02-02 by mefistotelis