kplindegaard/smbus2 #64: How to get i2c adapter capabilities

daylanKifky, 2021-01-25:
Hi! I'm trying to get the adapter capabilities, like stated in the README:
- Get i2c capabilities (I2C_FUNCS)
But I can't seem to the way to do so. Am I missing something?

Thanks!

daylanKifky, 2021-01-25:
I think I got it.
I should use `smbus2.SMBus._get_funcs()` combined with the `smbus2.I2cFunc` Intflag.

So for example to test 10 bit address capabilities I did:

```
dev = smbus2.SMBus(1)
has_10_bit_cap = dev._get_funcs() & smbus2.I2cFunc.ADDR_10BIT
```
Or perhaps there's a more straight forward way

kplindegaard, 2021-02-02:
Pretty much it but you can use the `.funcs` property as an alternative. Partly explained in the docs for `class I2cFunc`.

```python
bus = smbus2.SMBus(1)
has_10_bit_cap = bus.funcs & smbus2.I2cFunc.ADDR_10BIT
```

daylanKifky, 2021-02-03:
thanks for your answer

closed 2021-02-03 by daylanKifky