kplindegaard/smbus2 #66: FreeBSD support

tykling, 2021-02-01:
Hello :)

I would love to see FreeBSD support in smbus2 so Python i2c stuff would be easier in FreeBSD.

Before I do too much I wanted to ask if you are interested in accepting a PR for FreeBSD support (if I get that far)?

I've dug up https://github.com/freebsd/freebsd-src/blob/main/sys/dev/iicbus/iic.h which is used by https://github.com/freebsd/freebsd-src/blob/098dbd7ff7f3da9dda03802cdb2d8755f816eada/usr.sbin/i2c/i2c.c - the latter works well for communicating with i2c devices from the console. So I think the basics are in place.

I am not sure how to continue from here though, any pointers would be much appreciated! My ultimate goal is getting FreeBSD support in the Qwiic_py package for SparkFuns QWIIC ecosystem of sensors.

Thank you! :+1:

kplindegaard, 2021-02-02:
Hi @tykling. Pardon my lack of experience with FreeBSD.

Let's leave SMBus support out of the discussion for now and concentrate on the pure i2c capabilities. The smbus2 lib as well as the example you linked to communicate in similar ways over i2c:
* Calls are routed through `ioctl`.
* The payloads passed to `ioctl` are the same:
  * `class i2c_msg` in smbus2 is identical to `struct iic_msg`.
  * The collection/helper `class i2c_rdwr_ioctl_data` in smbus is identical to `struct iic_rdwr_data`.

That indicated to me that it should be possible to smbus2's `i2c_rdwr` right out of the box.... You know, the examples shown in the [I2C section in the readme](https://github.com/kplindegaard/smbus2/blob/master/README.md#i2c). Have you tested those?

tykling, 2021-02-08:
Hello :) Thank you for the input, it is much appreciated!

I've been trying to make this work. My i2c device is on address 48 and working well:

```
[root@pizero1 ~]# i2c -sv
dev: /dev/iic0, addr: 0x0, r/w: r, offset: 0x00, width: 8, count: 1
Hardware may not support START/STOP scanning; trying less-reliable read method.
Scanning I2C devices on /dev/iic0: 48 60 
[root@pizero1 ~]# i2c -a 48 -d r -c 2 -m tr
0c 19 
[root@pizero1 ~]# 
```

I am using the first i2c example from the README:
```
[root@pizero1 ~/smbus2]# cat i2ctest.py 
from smbus2 import SMBus, i2c_msg

with SMBus(1) as bus:
    msg = i2c_msg.read(48, 2)
    bus.i2c_rdwr(msg)
[root@pizero1 ~/smbus2]# 
```

With no changes to smbus2 I get the following error - which is understandable since it is using a Linux ioctl:
```
(smbus2_venv) [root@pizero1 ~/smbus2]# python i2ctest.py 
Traceback (most recent call last):
  File "i2ctest.py", line 5, in <module>
    bus.i2c_rdwr(msg)
  File "/root/smbus2/smbus2/smbus2.py", line 663, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 25] Inappropriate ioctl for device
(smbus2_venv) [root@pizero1 ~/smbus2]# 
```

It took me a while to resolve the correct ioctl macro I2CRDWR on FreeBSD - finally found the integer with help from a friend and this little program:

```
[root@pizero1 ~]# cat resolvemacro.c 
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>

typedef unsigned char u_char;

#include <dev/iicbus/iic.h>

int main(int argc, char *argv[])
{
        printf("0x%lx\n", I2CRDWR);
        return EXIT_SUCCESS;
}
[root@pizero1 ~]# cc -Wall -O2 resolvemacro.c 
[root@pizero1 ~]# ./a.out 
0x80086906
[root@pizero1 ~]# 
```

After putting the value 0x80086906 in this line: https://github.com/kplindegaard/smbus2/blob/master/smbus2/smbus2.py#L33 the error changes, so I think the ioctl is correct:
```
(smbus2_venv) [root@pizero1 ~/smbus2]# python i2ctest.py 
Traceback (most recent call last):
  File "i2ctest.py", line 5, in <module>
    bus.i2c_rdwr(msg)
  File "/root/smbus2/smbus2/smbus2.py", line 663, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 5] Input/output error
(smbus2_venv) [root@pizero1 ~/smbus2]# 
```

Something is still wrong though.

This is how far I got in the weekend. I will keep at it, but if you have any ideas I am all ears :)

Thanks! :+1:

tykling, 2021-02-08:
OK, playing a bit more with it, I appear to have discovered that it does work, but on some unexpected addresses. Changing my test script to loop over 0-255 and reading from each turns up my sensor on address 48, but it responds on 144 and 145:

```
(smbus2_venv) [root@pizero1 ~/smbus2]# cat i2ctest.py 
from smbus2 import SMBus, i2c_msg

with SMBus("/dev/iic0") as bus:
    for i in range(0,255):
        msg = i2c_msg.read(i, 2)
        try:
            print(f"reading from {i}")
            bus.i2c_rdwr(msg)
            for value in msg:
                print(value)
        except:
            #print("fail")
            pass
(smbus2_venv) [root@pizero1 ~/smbus2]# python i2ctest.py 
reading from 0
reading from 1
reading from 2
reading from 3
reading from 4
reading from 5
reading from 6
reading from 7
reading from 8
reading from 9
reading from 10
reading from 11
reading from 12
reading from 13
reading from 14
reading from 15
reading from 16
reading from 17
reading from 18
reading from 19
reading from 20
reading from 21
reading from 22
reading from 23
reading from 24
reading from 25
reading from 26
reading from 27
reading from 28
reading from 29
reading from 30
reading from 31
reading from 32
reading from 33
reading from 34
reading from 35
reading from 36
reading from 37
reading from 38
reading from 39
reading from 40
reading from 41
reading from 42
reading from 43
reading from 44
reading from 45
reading from 46
reading from 47
reading from 48
reading from 49
reading from 50
reading from 51
reading from 52
reading from 53
reading from 54
reading from 55
reading from 56
reading from 57
reading from 58
reading from 59
reading from 60
reading from 61
reading from 62
reading from 63
reading from 64
reading from 65
reading from 66
reading from 67
reading from 68
reading from 69
reading from 70
reading from 71
reading from 72
reading from 73
reading from 74
reading from 75
reading from 76
reading from 77
reading from 78
reading from 79
reading from 80
reading from 81
reading from 82
reading from 83
reading from 84
reading from 85
reading from 86
reading from 87
reading from 88
reading from 89
reading from 90
reading from 91
reading from 92
reading from 93
reading from 94
reading from 95
reading from 96
reading from 97
reading from 98
reading from 99
reading from 100
reading from 101
reading from 102
reading from 103
reading from 104
reading from 105
reading from 106
reading from 107
reading from 108
reading from 109
reading from 110
reading from 111
reading from 112
reading from 113
reading from 114
reading from 115
reading from 116
reading from 117
reading from 118
reading from 119
reading from 120
reading from 121
reading from 122
reading from 123
reading from 124
reading from 125
reading from 126
reading from 127
reading from 128
reading from 129
reading from 130
reading from 131
reading from 132
reading from 133
reading from 134
reading from 135
reading from 136
reading from 137
reading from 138
reading from 139
reading from 140
reading from 141
reading from 142
reading from 143
reading from 144
12
53
reading from 145
12
53
reading from 146
reading from 147
reading from 148
reading from 149
reading from 150
reading from 151
reading from 152
reading from 153
reading from 154
reading from 155
reading from 156
reading from 157
reading from 158
reading from 159
reading from 160
reading from 161
reading from 162
reading from 163
reading from 164
reading from 165
reading from 166
reading from 167
reading from 168
reading from 169
reading from 170
reading from 171
reading from 172
reading from 173
reading from 174
reading from 175
reading from 176
reading from 177
reading from 178
reading from 179
reading from 180
reading from 181
reading from 182
reading from 183
reading from 184
reading from 185
reading from 186
reading from 187
reading from 188
reading from 189
reading from 190
reading from 191
reading from 192
0
0
reading from 193
0
0
reading from 194
reading from 195
reading from 196
reading from 197
reading from 198
reading from 199
reading from 200
reading from 201
reading from 202
reading from 203
reading from 204
reading from 205
reading from 206
reading from 207
reading from 208
reading from 209
reading from 210
reading from 211
reading from 212
reading from 213
reading from 214
reading from 215
reading from 216
reading from 217
reading from 218
reading from 219
reading from 220
reading from 221
reading from 222
reading from 223
reading from 224
reading from 225
reading from 226
reading from 227
reading from 228
reading from 229
reading from 230
reading from 231
reading from 232
reading from 233
reading from 234
reading from 235
reading from 236
reading from 237
reading from 238
reading from 239
reading from 240
reading from 241
reading from 242
reading from 243
reading from 244
reading from 245
reading from 246
reading from 247
reading from 248
reading from 249
reading from 250
reading from 251
reading from 252
reading from 253
reading from 254
```
Great, finally some progress \o/ :+1: Can you spot why:

1. The two sensors I have connected on addresses 48 and 60 appear to be responding on different addresses when doing it through smbus2? The sensor on 48 is responding on 144 and 145, the sensor on 60 is responding on 192 and 193.
2. Each sensor appears to be active on two different addresses?

Thanks again!

unrelentingtech, 2021-02-16:
> Is this somehow related to the read/write bit being "counted"?

Yes, you're not supposed to pass raw addresses anywhere, the kernel always works with `(addr << 1) | rw`. See [iic(4)](https://www.freebsd.org/cgi/man.cgi?iic).

It's a bit confusing that you mixed hex and dec in the comment ;) but yeah, `0x60 << 1 == 192`. With your code how it is, it essentially takes `(addr << 1) | rw` from the user instead of taking `addr`.

BTW, SMBus is a completely separate API on completely separate devices, see [smb(4)](https://www.freebsd.org/cgi/man.cgi?smb). The `iicsmb` driver can do SMBus on top of I2C controllers, but note that there is no correlation guaranteed between `/dev/iicN` and `/dev/smbN` device numbers. So the Linux thing of supporting both I2C and SMBus API on the same `/dev/i2c-N` device won't work :/

tykling, 2021-05-13:
Hello :) I haven't forgotten or given up on this. I took over my new house on february 15th and haven't had much time since. I will get back to it asap!

brd, 2021-11-16:
@tykling I hope you are enjoying the new house :) I am wondering if you have had a chance to get back to this as I am interested as well.

0xc0decafe, 2022-05-31:
As I was missing this as well and had some spare time, I've been so bold and created a PR implementing support for FreeBSD. Tested on a Raspberry PI 3B. Have fun!