kplindegaard/smbus2 #67: Max block length in SMBus vs I2C spec, vs smbus2 implementation

mefistotelis, 2021-02-10:
SMBus spec 2.0:

> 5.5.7. Block write/read
> [...] The byte count does not include the PEC byte. The byte count may not be 0. A Block Read or Write is allowed to transfer a maximum of 32 data bytes.

The Figure 5-17 below defines "Byte Count" separately from "Data Byte" fields - so count is not included in the limit.

Smbus spec 3.1:

> 6.5.7 Block Write/Read
> [...] The byte count does not include the PEC byte. The byte count may be 0. A Block Read or Block Write is allowed to transfer a maximum of 255 data bytes.

The Figure 35 below defines "Byte Count" separately from "Data Byte" fields - so count is not included in the limit.

I2C spec rev. 6 - 4 April 2014:

I2C does not impose any restrictions on message length, other than stating that "I2C transmissions are sequences of 8-bit bytes".

---

Q: Is this library intended to meet SMBus 2.0 spec, or the latest one?

kplindegaard, 2021-02-11:
The smbus2 library is just a wrapper on top of `ioctl` and the i2c/smbus structs in the Linux kernel. The 32 bit limitation comes from there. See e.g. [https://www.kernel.org/doc/html/v5.4/i2c/smbus-protocol.html](https://www.kernel.org/doc/html/v5.4/i2c/smbus-protocol.html) and it's completely as the mercy of that + at the mercy of your driver.

I guess your question then becomes "Does Linux meet the SMBus 3.1 spec" and it doesn't seem to be the case.

mefistotelis, 2021-02-11:
You're right. It's a matter of improving the kernel module, which currently limits the size:
https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/i2c.h#L134

But if the kernel was patched, smbus2 imposes its own restrictions on message size:
https://github.com/kplindegaard/smbus2/blob/master/smbus2/smbus2.py#L50

kplindegaard, 2021-02-11:
True. That restriction must be lifted here too.