kplindegaard/smbus2 #73: import smbus2 error 'No module named 'fcntl'

budryerson, 2021-05-12:
I am writing a module for a Lidar device that uses the I2C interface.  I wrote my module importing the 'smbus' module. But when I created a distributable package that included the 'smbus' module as a requirement, it failed to install 'smbus' with about twenty lines of serious looking red text that began:  `ERROR: Command errored out with exit status 1:` etc.

Intimidated by this message, I rewrote my module to use 'smbus2' instead.   'smbus2' installed without error, but when I tried to import 'smbus2' (or 'SMBus' from 'smbus2'), I got this:
```
Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from smbus2 import SMBus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Bud Ryerson\Desktop\package\myve\lib\site-packages\smbus2\__init__.py", line 23, in <module>
    from .smbus2 import SMBus, i2c_msg, I2cFunc  # noqa: F401
  File "C:\Users\Bud Ryerson\Desktop\package\myve\lib\site-packages\smbus2\smbus2.py", line 25, in <module>
    from fcntl import ioctl
ModuleNotFoundError: No module named 'fcntl'
```
It seems that 'smbus2' cannot find 'fcntl' or maybe 'ioctl'.  Now it seems that I can't use 'smbus' and I can't use 'smbus2' either.  This is very frustrating.  Any advice would be appreciated.

Bud Ryerson
San Francisco

kplindegaard, 2021-05-15:
Hello Bud.
smbus2 is a library with Linux as its primary target OS. It uses Linux kernel structs and calls, so a Windows port is not straight-forward. I don't have much experience with i2c on Windows, I'm afraid. Sorry.

budryerson, 2021-05-15:
Thank you.  I don't like the answer, but I appreciate your thoughtful reply.
Bud Ryerson
San Francisco

closed 2021-05-15 by budryerson