kplindegaard/smbus2 #74: Find GPIO pins assigned to I2C bus number

deepcoder, 2021-05-17:
This is not a bug in smbus2, rather looking for some help figuring out how the I2C bus ordinal is mapped to physical I2C busses.

I have a Intel master device, the SEEDStudio Odyssey motherboard, it appears to have 8 busses according to sudo i2cdetect . Three of these are attached to a 'Synopsys DesignWare I2C adapter' and of these three, two are exposed on a 40 pin RPI compatible GPIO connector. The pins these two busses are attached to are fixed and defined.

However, on each boot, the I2C channel number used by smbus2 and other I2C api software seems to randomly change. 

I am wondering if someone has found a way to drill down through Linux /sys or /dev or ??? and figure out what pins are mapped to what I2C channel?

I have seen similar behavior in Linux RAID drivers, where the /dev/mdmXXX device id number changes on boot based on how fast the drives come on line in each RAID unit.

The problem in this particular motherboard may be exacerbated by the fact the BIOS does not assign a unique 'name' to each of the I2C busses managed by the  'Synopsys DesignWare I2C adapter' , all three report they name as 'Synopsys DesignWare I2C adapter', it seems other implementations add some unique identifier into the name : 

```
user@ODYSSEY-X86J4125:~$ sudo i2cdetect -l
i2c-3	smbus     	SMBus I801 adapter at f040      	SMBus adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-6	i2c       	i915 gmbus misc                 	I2C adapter
i2c-4	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-7	i2c       	AUX C/port C                    	I2C adapter
i2c-5	i2c       	i915 gmbus dpc                  	I2C adapter
```

Thank you for your work on this project, very helpful. And any insight you might have on this question.

kplindegaard, 2021-05-28:
Interesting problem. Because the first about unpredictable bus channel assignment triggers the second related to identifying which is which of the adapter carrying the same name. So, fixing the first eliminates the second too. 

Afraid I don't how much experience to draw from personally. Anyone else stopping by here who has?

sliedes, 2024-05-22:
I know this is late, but I fought with the same Odyssey board, and I believe I have an answer to this.

You can distinguish the devices by their address or probably ACPI path, accessible under /sys:

```
$ cat /sys/bus/i2c/devices/i2c-6/firmware_node/adr 
0x00170001
$ cat /sys/bus/i2c/devices/i2c-5/firmware_node/adr 
0x00170000
```

I _think_ these addresses should not change, and probably also not the ACPI paths (but that I think is up to the BIOS):

```
$ cat /sys/bus/i2c/devices/i2c-6/firmware_node/path 
\_SB_.PCI0.I2C5
$ cat /sys/bus/i2c/devices/i2c-5/firmware_node/path 
\_SB_.PCI0.I2C4
```