kplindegaard/smbus2 #85: Trouble reading and writing data vial I2C using smbus2 library

satya-prakash-7, 2022-07-08:
I'm trying to connect my energy measurement IC with Raspberry Pi 3B using the I2C protocol. I'm using `sudo i2cdetect -y 1` to detect the connected devices and it is correctly showing the address(7-bits) of the connected IC. The IC contains various registers of 8,16,24 and 32-bits. The register addresses are 16-bits long and I'm using the following code to access a register `0x021C(24-bits)` and write registers `0xFE(8-bit) and 0x0120(16-bit)`.

**Python Code**

```
from smbus2 import SMBus

bus = SMBus(1)

address = 0x38

bus.write_byte_data(address,0xFE,0xAD)  #Writing single byte of data
bus.write_i2c_block_data(address,0x120,[0x00,0x30])  #Writing multiple bytes of data to the register
block = bus.read_i2c_block_data(address,0x21C,3)  #Reading multiple bytes of data from the register
print(block)

bus.close()
```

The output should be` [0x8D,0xXX,0xXX] `but the code is producing output as `[0,0,0]`

- **Question-1**: Is giving a 16-bit address allowed in this Library or not?
- **Question-2:** If allowed then why the code is not working?

kplindegaard, 2022-08-03:
Answer: smbus2 is just a wrapper for i2c and SMBus support in the kernel. Hence, 16 bit addressing is not possible as per the definition of [i2c_smbus_ioctl_data](https://github.com/torvalds/linux/blob/e2b542100719a93f8cdf6d90185410d38a57a4c1/include/uapi/linux/i2c-dev.h#L44) struct in the kernel.

silver2row, 2022-10-09:
Hello,

Is there a way to use this library to interface a 32-bit armhf machine w/ a PCA9685 at a specific i2c address?

Seth