kplindegaard/smbus2 #90: ```PEC```-option when using the ```i2c_write()```-function?

Florian1FRANK, 2023-05-30:
Hello,

Does ```smbus2``` supports the ```PEC```-option when using the ```i2c_write()```-function as in the example below?
From my experience (with version ```0.4.2```), it seems the ```PEC```-option is not effective for the ```i2c_write()```-function, do you confirm?

Example:
```python
write = i2c_msg.write(80, [40, 50])
read = i2c_msg.read(80, 2)
with SMBus(1) as bus:
    bus.pec = 1
    bus.i2c_rdwr(write, read)
```

kplindegaard, 2023-06-17:
Remember that PEC is an SMBus protocol feature while `i2c_rdwr` is just pure i2c communication. I.e. no PEC for that command.

Florian1FRANK, 2023-06-19:
Thank you for the clarification. 
Would it be a valuable feature to have the possibility to enable the PEC for ```i2c_rdwr()``` as well?

Recently I used a chip requiring to use a I2C-message-structure + a trailing PEC byte ; for simple read actions I could use right away the smbus ```byte_read``` + in-built PEC feature of ```smbus2```, yet for write sequences I had to use ```i2c_rdwr()``` + a self coded PEC-byte. 
Hence my question.

kplindegaard, 2023-06-20:
Thanks for explaining. Interesting case and glad you pieced it together 😃

I have, however, tried to keep the lib as simple as possible and kept additional utilities to a minimum. My recommendation would thus be to to not include it and that particular device specific variations are kept outside of the lib itself. Sorry, but I hope you understand.

Florian1FRANK, 2023-06-21:
I think my original question is clearly answered, and the 'Issue' can be closed gracefully. 

Indeed keeping it simple is enough.

Maybe the documentation could gently remind the user that the inbuilt PEC feature is obviously *not* effective when invoking the ```i2c_rdwr()```-function?

closed 2023-06-21 by Florian1FRANK