kplindegaard/smbus2 #110: [Errno 121] Remote I/O error - AHT25 Sensor

DrKlipper, 2024-05-06:
Hi !

I try to write some python code for reading the AHT25 sensor. 
The sensor is nothing special - only a humidity / temp sensor. 

It´s connected to the Pi 4 directly with 15cm of wire cable to the I2C port of the Pi (Pin 3, 5). 

If I use i2cdetect I can see the sensor: 
```
pi@Pi4Test:~/python_sensor_aht20 $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Using i2ctransfer works also with no problems: 
```
pi@Pi4Test:~/python_sensor_aht20 $ i2ctransfer -y 1 w3@0x38 0xac 0x33 0x00 r6
0x98 0x74 0xeb 0x35 0xe4 0x92
```
And if I decode the resulting bytes I get plausible readings. I even can see that the temp rises when I touch the sensor. 

Now I tried to read the sensor with your python lib but without any luck ... 
I always got errors like this: 
```
    i2cBus.write_i2c_block_data(AHT20_I2CADDR, 0x0, AHT20_CMD_SOFTRESET)
  File "/home/pi/messy/.virtualenvs/Source/lib/python3.11/site-packages/smbus2/smbus2.py", line 643, in write_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error
```
Sometime I see Error 5, too (Input Output error)

I tried pullups with 2k2, 4k7 and 10k. But nothing seems to help. 

Do you have any idea what can cause this errors? It´s frustrating ... All other sensors I try work like a charm. But this sensor drives me crazy :-)

Hope on any hint. 

Dominik

DrKlipper, 2024-05-06:
Ok digging deeper I figured out the reading the device works. 
Only writing makes trouble. 

And if I look into the journal or dmeg I got this errors: 
```
[ 3629.067300] i2c i2c-3: sendbytes: NAK bailout.
[ 3658.804526] i2c i2c-3: sendbytes: NAK bailout.
[ 4325.794055] i2c i2c-3: sendbytes: NAK bailout.
[ 4350.696969] i2c i2c-3: sendbytes: NAK bailout.
```

Every write access produces one NAK bailout.

kplindegaard, 2024-07-11:
Hi @DrKlipper and thanks for your question. 

Judging from the `i2ctransfer` command you had, I suspect you want to send three bytes and subsequently read 6 bytes? Without knowing what how your source code looks, but from the error you included seems to indicate that you use the `write_i2c_block_data`...? I wouldn't use that for that scenario, though. You see, the `*_i2c_block_data` function require that your device has the concept of offset addresses, and I don't believe that is the case here. Hence, the wrong command.

What I would ask you to try is to use Example 6 in the readme as your base. You know, something like:

```python
from smbus2 import SMBus, i2c_msg

# Single transaction writing three bytes then read six
write = i2c_msg.write(0x38, [0xac, 0x33, 0x00])
read = i2c_msg.read(0x38, 6)
with SMBus(1) as bus:  # Don't know the i2c bus you use, so you may have to change this?
    bus.i2c_rdwr(write, read)
data = list(read)
print(data)
```

Very, very untested, but I would appreciate if you could try that :)

kplindegaard, 2025-12-28:
Excellent, @make-ing, and thanks for the confirmation. 

Yeah, let's close this one 😃

closed 2025-12-28 by kplindegaard